adafruit/Adafruit_ILI9341 #8: ESP8266 support

lunanigra, 2015-11-07:
Hello, is the Adafruit ILI9341 already compatible to ESP8266?

lunanigra, 2015-11-07:
Ok, seems the ESP8266 Arduino IDE already provides a modified lib... So, please just add a comment to the readme.

hb020, 2015-12-12:
Now that is no longer true. It was in staging, and did not make it in the 2.0 version. The version WAS in the arduino IDE git repo is here: https://github.com/Links2004/Adafruit_ILI9341 (and in sandeep's fork of the IDE)
And it is worth it. Fast, and complete (as far as I can tell). Can you please merge?

grahamehorner, 2016-01-13:
...\Adafruit_ILI9341\Adafruit_ILI9341.cpp: In constructor 'Adafruit_ILI9341::Adafruit_ILI9341(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)':

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:30:3: error: '_cs' was not declared in this scope

   _cs   = cs;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:31:3: error: '_dc' was not declared in this scope

   _dc   = dc;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:32:3: error: '_mosi' was not declared in this scope

   _mosi  = mosi;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:33:3: error: '_miso' was not declared in this scope

   _miso = miso;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:34:3: error: '_sclk' was not declared in this scope

   _sclk = sclk;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:35:3: error: '_rst' was not declared in this scope

   _rst  = rst;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp: In constructor 'Adafruit_ILI9341::Adafruit_ILI9341(int8_t, int8_t, int8_t)':

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:43:3: error: '_cs' was not declared in this scope

   _cs   = cs;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:44:3: error: '_dc' was not declared in this scope

   _dc   = dc;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:45:3: error: '_rst' was not declared in this scope

   _rst  = rst;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:47:3: error: '_mosi' was not declared in this scope

   _mosi  = _sclk = 0;

   ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:47:12: error: '_sclk' was not declared in this scope

   _mosi  = _sclk = 0;

            ^

[... 334 lines cut ...]
...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:582:17: error: '_cs' was not declared in this scope

    digitalWrite(_cs, LOW);

                 ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp: In member function 'uint8_t Adafruit_ILI9341::readcommand8(uint8_t, uint8_t)':

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:592:17: error: '_dc' was not declared in this scope

    digitalWrite(_dc, LOW); // command

                 ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:593:17: error: '_cs' was not declared in this scope

    digitalWrite(_cs, LOW);

                 ^

...\Adafruit_ILI9341\Adafruit_ILI9341.cpp:600:17: error: '_sclk' was not declared in this scope

    digitalWrite(_sclk, LOW);

                 ^

ladyada, 2016-02-08:
added in 
https://github.com/adafruit/Adafruit_ILI9341/commit/08167ad8740c5a4a43f771d539aa7886f5100719
use the default SPI pins SCK #14, MOSI #13, MISO #12 and CS #4, DC #5

closed 2016-02-08 by ladyada

alkhimey, 2016-02-20:
It seems that the latest commit have broken it again.

ladyada, 2016-02-20:
woops you're right. fixed!

rahulmr999, 2016-05-04:
I have some issues using the mentioned adafruit library and the ESP8266 arduino. I made the connections as mentioned 

> default SPI pins SCK #14, MOSI #13, MISO #12 and CS #4, DC #5.

The display works for a few seconds for the graphicsTest example. but the line test starts, it gets stuck and ESP reboots. Even when used in my project just to show some text, it reboots after a few successful attempts. 

I am using the 2.2 TFT ILI9341 LCD and ESPressoLite 1 which uses ESPWROOM 02

ladyada, 2016-05-04:
rahulmr999 this is not the place for tech support - contact the seller of your ESP module for support!