adafruit/Adafruit_ILI9341 #13: Examples will brick Arduino Zero & Fail to Program

rocketscream, 2016-04-20:
I have tested with a 2.8" ILI9341 LCD from EBay and hook them up to an Arduino Zero. Upon trying to upload the example sketches onto an Arduino Zero (native USB port), it will brick the USB native port. Even if the on-board EDBG is used to reprogram the bootloader, the native USB will no longer be detected by the PC (Windows 10 & Ubuntu 14.04). Before trying out on Arduino Zero, we tested them on our custom board based on Arduino Zero schematic on the native serial port. Thinking that it could be our design flaws, we tried on an original Arduino Zero and it ended as described as above. 

If we try to upload the code using the programming port of the Arduino Zero instead, we'll get the following error:

`** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
Warn : Adding extra erase range, 00000000 to 0x00001fff
wrote 40960 bytes from file /tmp/buildae05253274b53299ed1c1cf6c0dc729a.tmp/graphicstest.ino.bin in 5.396849s (7.412 KiB/s)
** Programming Finished **
** Verify Started **
Warn : target was in unknown state when halt was requested
Error: error executing cortex_m crc algorithm
Runtime Error: embedded:startup.tcl:454: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 498
at file "embedded:startup.tcl", line 454
the selected serial port at file "embedded:startup.tcl", line 454
 does not exist or your board is not connected`

Have you guys tested this on an Arduino Zero? It's best to provide some warning because it will brick many Arduino Zero (at least the native USB port will render useless after that) out there.

ladyada, 2016-04-20:
what happens when you double click the reset button while the native port is plugged into computer USB

rocketscream, 2016-04-20:
Yes, I tried that double clicking trick (before submitting this issue) and it remains not detected on the native USB port. If you can try it out on a Zero, it would great. You don't even need to hook up the ILI9341 LCD to make it happen. But, it will brick the Zero though.

ladyada, 2016-04-20:
just tried with a feather m0:

```
ILI9341 Test!
Display Power Mode: 0xFF
MADCTL Mode: 0xFF
Pixel Format: 0xFF
Image Format: 0xFF
Self Diagnostic: 0xFF
Benchmark                Time (microseconds)
Screen fill              2895905
Text                     208473
```

wasnt bricked...

ladyada, 2016-04-20:
you know that program Serial != native SerialUSB on Arduino Zero, right?

rocketscream, 2016-04-21:
That is very weird because it bricked 2 custom board of mine and a Zero. All using latest 1.6.8 IDE and 1.6.5 SAMD core. Tried both Atmel EDBG and SAM-ICE to reburn the bootloader, but it will never get recognized or detected on both Windows and Ubuntu (both are different machine). And yes, I know about the SerialUSB. The point is it no longer get detected as a native USB port after the sketch upload. What version of the software you are currently using? I did tried the hourly built one too.

ladyada, 2016-04-21:
im using 1.6.7 and the latest version of the adafruit samd boards pkg
closing this issue since i can't reproduce it and there isn't anything going on in the code that seems suspect, but plz keep posting if you have any insight :)

ladyada, 2016-04-21:
also, of course, we do not support EBay purchases of non-adafruit goods

you should talk to the seller - maybe they can help since they sold you the hardware, or perhaps they have working library code :)

closed 2016-04-21 by ladyada

rocketscream, 2016-04-21:
It's okay, I'll try to figure it out here. I didn't even plug the LCD as a start, so I would rule out the LCD as a start. Just loaded the code onto the Arduino Zero using the USB native port would brick it. Will update if there's anything I found.

rocketscream, 2016-04-21:
I can confirm that this is a bug related to the SAMD core. Submitted an issue [#137](https://github.com/arduino/ArduinoCore-samd/issues/137). It just so happen that the examples in this library (probably the size?) made it happen first try of using it with the latest 1.6.8 IDE and 1.6.5 SAMD core. Sorry for the wrong alarm. Thanks.

FrogManBird, 2017-01-05:
Successfully unbricked the Arduino Zero Pro Native port.  Many hours of blood, sweat and beers...  This is what worked for me:
1.	Plug in to programming port
2.	Set up the Arduino.cc 1.8.0 IDE as follows:
   a.	Board:  Arduino M0 Pro Programming Port
   b.	Port: same thing
   c.	Programmer:  Atmel EDBG
   d.	The com port (with proper driver) shows as EDBG Virtual Com Port
3.	Close Arduino
4.	Open Atmel Studios 7
  a.	Tools/Device Programming/Tool EDBG/Device ATSAMD21G18A/Interface SWD
  b.	Apply/Read the device info
  c.	Go to Fuses:
      i.  Change fuse register USER_WORD_0 to 0xD9FEC7FF  (Atmel default value)
     ii.  Change fuse register USER_WORD_1 to 0xFFFFFE5D  (Atmel default value)
    iii.	  Click Program
5.	Close Atmel Studios 7
6.	Power cycle board
7.	Open Arduino.cc IDE
8.	Plug into programming port and burn bootloader
9.	Now when plugging into native port, com port detects Arduino M0 PRO Native Port
10.	Note that when Atmel Studios is opened up again USER_WORD_1 stays to what was set, but USER_WORD_0 has changed to 0xD9FEC7F9 (last digit changed from F to 9).  That’s what the bootloader burned in and now it’s correct (don’t change it back).

ianrrees, 2021-05-24:
In case someone else stumbles on to a similar problem and uses openocd, I've successfully used the invocation below, followed by a hard reset, to restore the NVM user row.

`openocd -f interface/ftdi/dp_busblaster_kt-link.cfg -f feather.cfg -c init -c "reset halt" -c "at91samd nvmuserrow 0xFFFFFC5DD8E0C7FA"`

The `-f interface/ftdi/dp_busblaster_kt-link.cfg` refers to the particular JTAG/SWD probe., and feather.cfg contains:
```
transport select swd
set CHIPNAME at91samd21g18
set CPUTAPID 0x0bc11477
source [find target/at91samdXX.cfg]

adapter srst delay 100
adapter srst pulse_width 100
```