adafruit/Adafruit_ILI9341 #14: Issue with running the example 'graphicstest'

jdlowman2, 2016-04-25:
This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
Arduino: 1.0.6 (Windows NT (unknown)), Board: "Arduino Uno"
graphicstest.ino: In function 'long unsigned int testFillScreen()':
graphicstest:113: error: 'yield' was not declared in this scope
graphicstest.ino: In function 'long unsigned int testLines(uint16_t)':
graphicstest:159: error: 'yield' was not declared in this scope
graphicstest.ino: In function 'long unsigned int testFilledRects(uint16_t, uint16_t)':
graphicstest:255: error: 'yield' was not declared in this scope
graphicstest.ino: In function 'long unsigned int testFilledTriangles()':
graphicstest:327: error: 'yield' was not declared in this scope
graphicstest.ino: In function 'long unsigned int testFilledRoundRects()':
graphicstest:361: error: 'yield' was not declared in this scope

ladyada, 2016-04-25:
you must use arduino 1.6+

closed 2016-04-25 by ladyada

harvitronix, 2016-09-10:
I had this issue trying to compile on Ubuntu.

In case others find this with the same problem: the packages installed with apt are 1.0.6. Remove those and manually download 1.6 from the Arduino website. Extra, run Arduino and the graphictest (and other examples) will work great!