adafruit/Adafruit_ILI9341 #15: Unable to compile examples for NodeMCU 0.9 board

papaiatis, 2016-10-16:
I downloaded the latest version (1.0.1) of Adafruit ILI9341 library and set my board to NodeMCU 0.9 (ESP-12 Module). I'm using Arduino IDE 1.6.12.

I wanted to upload the graphicstest example to my NodeMCU device but during compile I'm getting lots of error messages:

    Adafruit_ILI9341.cpp:30:3: error: '_cs' was not declared in this scope
    Adafruit_ILI9341.cpp:31:3: error: '_dc' was not declared in this scope
    Adafruit_ILI9341.cpp:32:3: error: '_mosi' was not declared in this scope
    ...

If I set the board to Arduino UNO the examples are compilling just fine.
What do you think?

papaiatis, 2016-10-16:
Okay, so I came upon this website: http://www.nihamkin.com/2016/03/04/connecting-esp8266-with-ili9341-tft-display/ which says the ILI9341 library available through Arduino IDE Library Manager is broken and I should download it directly from this github repo. He was right!
Indeed the code here in Github works!
I'm closing this ticket then.

closed 2016-10-16 by papaiatis

ladyada, 2016-10-17:
version bumped in IDE manager, to 1.02