adafruit/Adafruit_ILI9341 #18: __AVR__ optimizations in spiwrite, spiread don't work on Arduino Uno Wifi

cornwarecjp, 2016-11-07:
- Arduino board:  Arduino Uno Wifi
(note: board made by by ardiuno.org)

- Arduino IDE version (found in Arduino -> About Arduino menu):  2:1.0.5+dfsg2-4
(note: IDE made by arduino.cc, distributed by Debian Jessie. It doesn't have support for the Arduino Uno Wifi, so I selected Arduino Uno in the IDE. I copied the Wifi library from the arduino.org IDE.)

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
* I re-flashed the Wifi chip on the Arduino with self-made firmware. I think this step is irrelevant, since the Wifi chip isn't connected to the Arduino SPI, and my self-made firmware isn't supposed to do anything anyway.
* I attached the 2.8" TFT Touch shield to the Arduino (box refers to http://adafru.it/1651, and PCB says it's a v2.0 shield)
* I installed the Adafruit_GFX, Adafruit_ILI9341 and Adafruit_STMPE610 libraries
* I tested the Adafruit_STMPE610/TouchTest example; it worked after modifying it a bit (had to use hardware SPI with CS 8 instead of I2C)
* I tested the Adafruit_ILI9341/graphicstest example. This didn't work:
  - it printed "ILI9341 Test!"
  - then: nothing (no printing, no screen changes)

After some debugging, I found that it started working by disabling the __AVR__ specific code in the spiwrite and spiread functions in Adafruit_ILI9341.cpp. So, in both functions, where the original code says
#if defined (__AVR__)
my modified code says
#if defined (DISABLED) && defined (__AVR__)

With this change, I force that the generic SPI.transfer(..) code is used instead of the __AVR__ specific code in the library. Somehow, that fixed the problem: the graphicstest example works, and the touchpaint example also works, so I assume I now have a completely working touch shield.

I don't know if this problem is specific to my situation, or if other people are going to run into it. In the second case, you might use this report to improve your library. If you know why my system has this problem, please let me know: maybe it could be beneficial to me to use the __AVR__ specific code if there is some way to do that (I assume it's there for some reason - maybe an optimization?).

driverblock, 2016-11-10:
I don't know why it's like that.  AVR should be using SPI Transfer if it's available.  I'll look into it. Thanks for the tip!

ladyada, 2018-08-11:
hiya just closing some old open PR's
we've updated the library to be based off of Adafruit_SPITFT which abstracts the SPI stuff away. please try it again, if its still being weird, please open another issue!

closed 2018-08-11 by ladyada