adafruit/Adafruit_ILI9341 #29: Please re-add pushColors (plural of pushColor)

osos, 2017-05-24:
It seems that during the refactoring and optimization of the code, the pushColors functions was removed.

I see that the pushColor was re-added later, but I have some code relaying on the former pushColors (plural) which in turn could also take into consideration if data alignment has big/littleEndian.

Unfortunately I am not sure how to re-add the function myself.

PaintYourDragon, 2017-05-24:
Suggest using writePixels() instead, which is part of the newer SPI-transaction-based code. You'll need to start/end the transaction, but otherwise the address window and write operation are similar to before, something like:
```
display.startWrite();
display.setAddrWindow(...);
display.writePixels(...);
display.endWrite();
```

closed 2017-05-24 by PaintYourDragon

ideaalab, 2020-05-17:
We needed pushColors for a project from a couple of years ago. Solved it with this:
```
void Adafruit_ILI9341::pushColors(void * colorBuffer, uint16_t nr_pixels, uint8_t async){
  if (hwSPI) spi_begin();
  
  *dcport |=  dcpinmask;
  *csport &= ~cspinmask;

  if (async==0) {
    SPI.dmaSend(colorBuffer, nr_pixels, 1);
    *csport |= cspinmask;
  } else {
    SPI.dmaSendAsync(colorBuffer, nr_pixels, 1);
  }

  if (hwSPI) spi_end();
```
Dont know if its ok, but the project is working fine.