adafruit/Adafruit_ILI9341 #32: ESP8266, most recent version does not work, 1.0.2 is ok.

Seppeltronics, 2017-09-15:
Hello, I’m using an ESP8266 and a ILI9341, with the 1.0.2 it just works fine, with the most recent version it fails, just shows a white screen. I use the 2.0.3 SPI Driver and Adafruit_GFX from May.27. 2017 . My guess is that the tft.begin(), which was changed may cause the issiue,...  .

I get a Stacktrace:

<<<stack<<<
ÆU)�
ôŒ

ÿ(Š**š¢Š¢•ÍÑ…

Soft WDT reset

ctx: cont 
sp: 3ffef270 end: 3ffef4d0 offset: 01b0

>>>stack>>>
3ffef420:  0000013f 000000ef 3ffee3ac 4020246f  
3ffef430:  00000000 3ffee460 000018f0 40202b06  
3ffef440:  feefeffe 00000001 3ffee3ac 00000140  
3ffef450:  00012c00 00000000 3ffee3ac 40202bd5  
3ffef460:  000000f0 00000000 3ffee3ac 4020392c  
3ffef470:  00000000 00000000 3ffee3ac 4020246f  
3ffef480:  3fffdad0 00000028 3ffee3ac 3ffee4a8  
3ffef490:  3fffdad0 00000000 3ffee3ac 40203c59  
3ffef4a0:  3fffdad0 00000000 3ffee3ac 40201c49  
3ffef4b0:  feefeffe feefeffe 3ffee4a0 402035a0  
3ffef4c0:  feefeffe feefeffe 3ffee4b0 40100718  
<<<stack<<<
R)�
ôŒ

ÿ(Š**š¢Š¢•ÍÑ…

------------------------------------------------------

Following pin’s are used:

ILI9341 	:ESP8266
D/C	:GPIO16
SCLK	:GPIO14(SCK)
MISO	:GPIO12(MISO)
MOSI	:GPIO13(MOSI)
CS	:GPIO0
RESET	:RESET

The Code used is:

#include <Arduino.h>
#include <stdint.h>
#include <SPI.h>
#include <Adafruit_ILI9341.h>
#include <Adafruit_GFX.h>

#define TFT_DC D2
#define TFT_CS D8

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

void setup()
{
  tft.begin(500000);
  tft.setRotation(1);
}

void loop()
{
  tft.fillScreen(ILI9341_BLACK);
  delay(500);
  tft.fillScreen(ILI9341_GREEN);
  delay(500);
}

ladyada, 2017-09-15:
1.0.10 just tested, works fine. use latest

closed 2017-09-15 by ladyada

Seppeltronics, 2017-09-16:
Actually it does not with 1.0.10 . I do not exactly know what is wrong, given the fact that it works with 1.0.2, something is different with 1.0.10. I used the "graphicstest.ino", all I modifies was:

 // For the Adafruit shield, these are the default.
//#define TFT_DC 9
//#define TFT_CS 10
#define TFT_DC D2
#define TFT_CS D8

Someone(WerecatF) https://github.com/adafruit/Adafruit_ILI9341/pull/12 mentioned GPIO0 may be an error in your Library, is that issue fixed already? I hooked up a Logic-Analyzer, from the brief look they are about the same, there seems to be a problem with the ChipSelect. "Something" is happening on GPIO0 and it does not seem to use the ESP8266es HW SPI?

[image]

Overview 1.0.2:
[image]

Overview 1.0.10:
[image]

ladyada, 2017-09-16:
try the featherwing example wiring, we used it with the touchpaint demo last night