adafruit/Adafruit_ILI9341 #42: Latest update (1.3.1) causes artifacts on display

merdok, 2019-03-04:
Hi,

with the latest update 1.3.1 the display will get artifacts (text are not readable anymore and don't update) after around 10/15 minutes of run time. After downgrading back to 1.2.0 everything is fine again. Could you please fix that?
My setup:
ESP32 + 2.4 Touchscreen (ILI9341, XPT2046)

PaintYourDragon, 2019-03-05:
Was Adafruit_GFX updated as well, or just the ILI9341 library?

PaintYourDragon, 2019-03-05:
Also, hardware SPI, or bitbanging?

merdok, 2019-03-05:
Yes, Adafruit_GFX was also updated. But with the latest Adafruit_GFX and Adafruit_ILI9341 1.2.0 it is working. When i update Adafruit_ILI9341 to 1.3.1 then the issue starts to appear.

I suppose hardware SPI... Here is my init method:
`Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);`

PaintYourDragon, 2019-03-06:
Haven't been able to reproduce the problem yet…been running overnight without glitching. In order to help track down the cause of what you're seeing, here's some things to try...

- Does the problem occur consistently between 10-15-ish minutes?
- Is it possible that rather than being time-related, it's something in the nature of the text being displayed? Printing any unusual characters around that time, or with characters clipping off the screen?
- Is this wired up through a breadboard, or something more permanent?
- If breadboard, try throttling back the SPI data rate…this can be passed as an argument to the ILI9341 begin() function, e.g. 8 MHz like this:
`tft.begin(8000000);`

merdok, 2019-03-06:
- I did a couple of more tries with the latest update and the issues comes up randomly (as it seems), sometimes after 10-15 minutes but also sometimes after 60 minutes.
- I do not print any unusual characters and the characters are not clipping off screen. During all the time there are always the same characters on screen.
- I actually use the Lolin D32 Pro board and are connecting the 2.4 touch shield from Lolin using the display connector (which is as far as i know nothing else then hardware SPI), but like i said the same setup is working fine on 1.2.0 of this library.

I have attached some pictures so you can see what is happening. The first picture shows how it should look like at all times. The clock in the middle updates every second, the values above and below the clock update every 60 seconds.
On picture 2 you can see the glitching. I marked all the sections and will blue lines and numbers: 
- On section 1 the values does not update anymore for some reason (I still see the values updating in the serial monitor, so i guess the text writing function is suddenly broken)
- On section 2 the clock is frozen (yellow numbers) but the clock is still updating (white numbers, but should be yellow) and they overlap.
- On section 3 all the values are also frozen and overlapping with new values. 
Picture shows what happens after the screen starts to glitch and i do a soft reboot(without pulling the power plug). The complete screen is suddenly missing most of the text, only the text which is overlapping in picture 2 is still present for some reason.

And here is the code which i use to write the text
```
    tft.setTextColor(ILI9341_WHITE, ILI9341_BLACK);
    tft.setTextSize(1);
    tft.print("Temperature: ");
    tft.setTextSize(1);
    tft.print(temp_value);
```

I hope that i could explain it more detailed :) And again (this is very important here) all of this is not happening with version 1.2.0 of this library, ONLY with 1.3.1!

[image]
[image]
[image]

Loimulohi, 2019-03-13:
I have same problem as merdok. Using Adafruit_ILI9341 v1.3.1 is causing artifacts on screen and sometimes screen is completely white and flickering and stays that way. Usually this happens few seconds after power on but sometimes it takes couple of minutes to happen.

I don't have any problems using ILI9341 v1.2.0 and GFX v1.3.2. I have not tried latest GFX and ILI9341 v1.2.0 is that combination working or not.

Using custom made ESP32 board and 2.8" TFT.

PaintYourDragon, 2019-03-16:
Please try GFX v1.4.4 and let me know if that clears it up. Thanks!

Loimulohi, 2019-03-19:
I have updated to GFX v1.4.6 and ILI9341 v1.3.4 and no glitches for 15 hours so far, so it seems to be working! Thank you very much!

merdok, 2019-03-19:
@PaintYourDragon 
Tried it today with GFX v1.4.7 and ILI9341 v1.3.4 and the issue is still there. As soon as i downgrade back the ILI9341 library to 1.2.0 and leave GFX at v1.4.7, everything is working again.

PaintYourDragon, 2019-03-20:
@merdok 
An earlier message suggested the problem started with ILI9341 version 1.3.1.
Is that not the case…is it actually a specific GFX release where the issue starts to appear?

merdok, 2019-03-20:
Doing some testing now to find out the specific versions of the packages. 
Tried iwith GFX v1.3.0 and ILI9341 v1.2.0 and everything is working fine. Will be going slowly up to and see at what version the issue will appear.

merdok, 2019-03-20:
Ok, did a lot of tests today and i can confirm once again that this issue is related to the ILI9341 library.
I installed the ILI9341 1.2.0 and after that i tested it with GFX library from 1.3.0 to 1.4.7 and everything was working fine. Then i tested ILI9341 1.3.1 and GFX 1.4.7 and after a few minutes the issues appeared.
So some of the changes which was made in 1.3.1 of the ILI9341 library is breaking it. This has nothing to do with the GFX library.

So with any version of ILI9341 library since 1.3.1 the artifacts starts to appear. No matter which GFX library version!

PaintYourDragon, 2019-03-21:
This is strange…the few changes at v1.3.1 should apply only to the ESP8266.

There was some stuff that changed in 1.3.0 that might be important though. Try changing the display’s begin() call by passing 40000000 (40 MHz) as an argument, e.g.
`tft.begin(40000000);`

merdok, 2019-03-21:
I have tried it with `tft.begin(40000000);` and it seems to work now. I guess that the default values are messed up since version 1.3.1. Could you please fix that?

PaintYourDragon, 2019-03-22:
"Classic" default SPI speed behavior has been restored in v1.3.6; should work as before now.

closed 2019-03-22 by PaintYourDragon

merdok, 2019-03-22:
Thanks for the fix! Can confirm that with version 1.3.6 everything works again!

PaintYourDragon, 2019-03-22:
Yay! Good to hear.

Curious though: are any of your sensors interfaced via SPI, and do they take readings during an interrupt, like a timer or data-ready signal?

merdok, 2019-03-22:
I have a BME680 connected over I2C and i read out the sensor every 60 seconds using the Adafruit_BME680 library.

Loimulohi, 2019-03-23:
I have BME280, EEPROM, DS3231 and LM75A connected to I2C bus and four DS18B20 1-wire temperature sensors and SD Card connected to SPI bus.