adafruit/Adafruit_ILI9341 #49: Adafruit_ILI9341::scrollTo uint16 vs uint8

marcmerlin, 2019-05-29:
while compiling with all warnings on:
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp:259:46: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     sendCommand(ILI9341_VSCRSADD, (uint8_t*) y, 2);

Sure enough, the function is
void Adafruit_ILI9341::scrollTo(uint16_t y) {
    sendCommand(ILI9341_VSCRSADD, (uint8_t*) y, 2);
}

Given that a TFT can be wider than 256 chars, uint16_t seems reasonable, but spiWrite(*dataBytes) may only allow a single byte, i.e. a scroll of up to 255 pixels?

If so, changing scrollTo to take uint8_t seems reasonable?

makermelissa, 2019-05-29:
This should probably make use of __builtin_bswap16(). I’ll have to take a closer look to be sure. sendCommand looks at the data in bytes, which is the reason for casting to a byte pointer.

marcmerlin, 2019-05-29:
understood, thanks for having a look (didn't look at that code as I'm not using it, just noticed the warning).
More generally, in your copious spare time :-D , may I recommend you set the arduino IDE to full warnings and look at the output when building your multiple libraries?
It will either identify some corner cases bugs, or allow you to fix spurious warnings that will show up for everyone using all warnings to find bugs in their own code (which is what I do).
Thanks.

ladyada, 2019-05-29:
hiya once we have all our libraries travis'd we can turn on -Wall 
at this time, we do not have the bandwidth to do so for all 300 libraries we maintain

marcmerlin, 2019-05-29:
totally understood, and OMG, 300 libraries already?  I knew you wrote a lot, but that's even more than I imagined. Obviously you've earned infinite karma with that level of OSS driver contributions.
Thanks again for all of those, some of which I use happily, a few I made meager contributions to. Cheers.

PR #50 (2019-06-01): Fixed compiler warning for scrollTo()
Fixes #49.
closed 2019-06-01 by makermelissa