adafruit/Adafruit_ILI9341 #52: git TOT library does not work with teensy 3.6

marcmerlin, 2019-06-01:
I have graphicstest working fine on ESP32.
I also have teensy 3.6 working fine with the same TFT using ILI9341_t3, but I'd rather not use it as it's a different and somewhat incompatible API.

I did init with
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
which uses software SPI and works fine on ESP32. With teensy 3.6 (arduino 1.8.8 + teensyduino-1.45), I get:
```
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.h:140:38: error: expected ')' before 'busWidth'
         Adafruit_ILI9341(tftBusWidth busWidth, int8_t d0, int8_t wr, int8_t dc,
                                      ^
Multiple libraries were found for "Adafruit_ILI9341.h"
 Used: /home/merlin/Arduino/libraries/Adafruit_ILI9341
 Not used: /var/local/arduino-1.8.8-teensyduino-1.45/hardware/teensy/avr/libraries/Adafruit_ILI9341
```

If I comment out this constructor line I'm not using in your library, I get:
```
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp: In constructor 'Adafruit_ILI9341::Adafruit_ILI9341(SPIClass*, int8_t, int8_t, int8_t)':
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp:122:77: error: invalid conversion from 'SPIClass*' to 'int8_t {aka signed char}' [-fpermissive]
   Adafruit_SPITFT(ILI9341_TFTWIDTH, ILI9341_TFTHEIGHT, spiClass, cs, dc, rst) {
                                                                             ^
In file included from /home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.h:43:0,
                 from /home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp:49:
/var/local/arduino-1.8.8-teensyduino-1.45/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.h:27:9: note:   initializing argument 3 of 'Adafruit_SPITFT::Adafruit_SPITFT(uint16_t, uint16_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)'
         Adafruit_SPITFT(uint16_t w, uint16_t h, int8_t _CS, int8_t _DC, int8_t _MOSI, int8_t _SCLK, int8_t _RST = -1, int8_t _MISO = -1);
         ^
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp: At global scope:
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp:141:35: error: expected constructor, destructor, or type conversion before '(' token
 Adafruit_ILI9341::Adafruit_ILI9341(tftBusWidth busWidth,
```

If I switch to the version in teensyduino, it works.
Looks like your recent changes made it incompatible with teensy?

makermelissa, 2019-06-01:
Hi, please try using the latest Adafruit _GFX library. The one included with Teensyduino is an older library.

marcmerlin, 2019-06-01:
whoops, you are correct.  Why would the arduino env pick the ILI lib from you and the old Adafruit::GFX lib from teensyduino which was much older (I had both) while happily taking the latest Adafruit_ILI9341 from the same stock arduino/libraries directory?
I deleted the teensyduino copy to stop it from getting one lib in one place and the other lib in the other place, and now things work again.
My apologies for missing that and wasting your time.

closed 2019-06-01 by marcmerlin

makermelissa, 2019-06-01:
It’s no problem. You aren’t the first to run into this issue.

ladyada, 2019-06-01:
@PaulStoffregen please remove or update Adafruit_GFX from the teensyduino installer