adafruit/Adafruit_ILI9341 #58: overwrite text

avandalen, 2019-10-25:
I have made an extension for overwriting numbers. See here:
How to overwrite text to the Arduino ILI9341 display:
https://www.avdweb.nl/arduino/hardware-interfacing/ili9341-tft
You can integrate it into your library if you want.

ladyada, 2019-10-25:
hiya a PR to SPITFT (in Adafruit_GFX) would be the best way to get it included! we cannot use code that is under a "CC Non Commercial" license

avandalen, 2019-10-26:
Hi

I changed all my software to MIT licence now

what do you with
hiya a PR to SPITFT 

Kind regards 
Albert

From: ladyada 
Sent: Friday, October 25, 2019 6:18 PM
To: adafruit/Adafruit_ILI9341 
Cc: Albert van Dalen ; Author 
Subject: Re: [adafruit/Adafruit_ILI9341] overwrite text (#58)

hiya a PR to SPITFT (in Adafruit_GFX) would be the best way to get it included! we cannot use code that is under a "CC Non Commercial" license

caternuson, 2019-11-14:
@avandalen In general:
* Fork the Adafruit_GFX repo: <https://github.com/adafruit/Adafruit-GFX-Library>
* Make a branch
* Add your code to the branch
* PR that branch back to the Adafruit_GFX repo

More info in this guide:
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github/overview

avandalen, 2019-11-17:
I accidentally verified my own branch (GPX.cpp and GPX.h), can you undo this?

avandalen, 2019-11-17:
Ingnore this previous comment: I accidentally verified my own branch (GPX.cpp and GPX.h), can you undo this?

closed 2019-11-17 by avandalen