adafruit/Adafruit_ILI9341 #59: Support for SPI Settings

Decee1, 2019-10-29:
- Arduino board:  Arduino Uno

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.9

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Use device that needs SPI_MODE1. Example: ADS1220
ADS1220 LIB: https://github.com/Protocentral/Protocentral_ADS1220 using modified tommz9 pull request

Hello.
I would like to request support for SPI settings as i got a ADS1220 which users spiMode1 and the ILI display uses SPImode0. This gives some problems and as of now Arduino also recommends using SPIsettings for libraries to ensure that it runs at the correct SPI settings

SPI settings reference:
https://www.arduino.cc/en/Reference/SPIbeginTransaction

ladyada, 2019-10-29:
this library uses SPI transactions

closed 2019-10-29 by ladyada

Decee1, 2019-10-29:
Ah my bad, i was running the older PDQ_ILI9340