adafruit/Adafruit_ILI9341 #62: scrollMargin() has a typo

prenticedavid, 2020-05-01:
scrollMargin() only works for bottom == 0
e.g.   // TFA+VSA+BFA must equal 320
VSA = 320 - TFA - BFA;  // is correct
VSA = 320 - TFA **+** BFA;  // only ok for 320 - TFA **+** 0

The same problem occurs with Adafruit_HX8357 library.
You don't implement scrollMargin() on Adafruit_ST7789 or Adafruit_ST7735

```
void Adafruit_ILI9341::setScrollMargins(uint16_t top, uint16_t bottom) {
  // TFA+VSA+BFA must equal 320
  if (top + bottom <= ILI9341_TFTHEIGHT) {
    uint16_t middle = ILI9341_TFTHEIGHT - top + bottom;
    uint8_t data[6];
    data[0] = top >> 8;
    data[1] = top & 0xff;
    data[2] = middle >> 8;
    data[3] = middle & 0xff;
    data[4] = bottom >> 8;
    data[5] = bottom & 0xff;
    sendCommand(ILI9341_VSCRDEF, (uint8_t *)data, 6);
  }
}
```

David.

KurtE, 2021-05-14:
@ladyada  - Ran into this when I was asked to add this to my ILI9341_t3n library.  

My fix is: uint16_t middle = ILI9341_TFTHEIGHT - (top + bottom);

ladyada, 2021-05-14:
@kurte thanks, please submit a PR :)