adafruit/Adafruit_ILI9341 #65: Hardware SPI don't work with ILI9341 - 3 bugs and 3 solutions here

electroremy, 2020-06-16:
Hi,

I'm use the Adafruit_ILI9341 and the GFX library to drive a ILI9341 TFT SPI touch screen on a Arduino UNO with an Ethernet Shield 2 (for the touch pad I use URTouch.h)

I've found some bugs and also solutions :

**1st :**
If you use
`Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST)`
you've got an error "call of overloaded 'Adafruit_ILI9341(int, int, int)' is ambiguous"
So you have to edit Adafruit_ILI9341.h and Adafruit_ILI9341.cpp to comment Adafruit_ILI9341() fonctions for ESP8266
I think you have to change function names to avoid this.

**2nd :**
ILI9341 use SPI MODE 3 and cannot work with the default SPI MODE the library uses with hardware SPI
Curiously, the software SPI use the good mode.
Thus, you have to edit Adafruit_ILI9341.cpp and replace
`initSPI(freq);`
by
`initSPI(freq, SPI_MODE3);`

**3rd**
When you have quite long wires (eg. 20cm dupont wire) and slow 5V/3.3V level converter (or just 10K resistors !) from the Arduino UNO and the TFT you've got a speed limitation.
Remember that in electronics :
- long wire = coil = low pass filter
- resistor + logical input capacitance = low pass filter
Indeed, in the sketch, you have to use
`tft.begin(3000000);`
instead of
`tft.begin();`

You can also use a fast SPI speed with short wires and a hi-speed level converter.

I found that a lot of people don't know these problems and use
`Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);`
with the hardware SPI pins...
that's very very bad...
that's very slow...
and it doesn't work with other SPI devices...

I wonder that the constructor have explicit names as "tftHardwareSPI" and "tftSoftwareSPI"

Please, can you update your library and also your documentation and sample codes ?

Bests regards.

Gerriko, 2023-09-03:
I also have a hardware SPI bug. Maybe this is bug 4.
I am using this library with an Arduino Giga R1. I was trying to get this library to work with SPI1 but was having problems. 

It works when I try the default SPI port using the following:

```
// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

```

But if I then add in the SPI Class it fails to work - I just get a dimmer white screen with very faint flicker (this is the same result if I use &SPI1):

`Adafruit_ILI9341 tft = Adafruit_ILI9341(&SPI, TFT_CS, TFT_DC);
`

Gerriko, 2023-09-04:
I discovered my issue. The library has DC and CS pins swapped, when using Adafruit_ILI9341(SPIClass *spiClass, int8_t dc, int8_t cs = -1, int8_t rst = -1);

FrankSchuby, 2024-04-23:
Hello, I'm currently working in software mode.
#define TFT_CS 54
#define TFT_DC 55
#define TFT_MOSI 56
#define TFT_CLK 57
#define TFT_RST 58
#define TFT_MISO 59
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
But now I want to set hardware SPI, where exactly do I set the SPI interface?

FrankSchuby, 2024-04-24:
Isn't there anyone here who could help me?
I want to use hardware SPI but don't know how to declare it in the code. I work with
Platformio
Adafruit GrandCentral M4 Samd51
There must be a way to specify the SPI port.
#define TFT_CS 54
#define TFT_DC 55
//Hardware SPI
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);