adafruit/Adafruit_ILI9341 #74: Arduino ILI9341 simulation 😀

matititam, 2021-09-22:
Dear Team,

Good day

We are ever grateful for such wonderful libraries. I have here a permanent link for the simulation.  
https://wokwi.com/arduino/projects/307567275170333248

It would be awesome if you can make this link available to the future readers.
We will be excited to hear about the feedback on the simulator as well. May we know, how we can make it better to publish such useful libraries in the future. 

Thank you!
Matititam

[image]

makermelissa, 2021-11-03:
Hi, I just got around to trying this out. That's pretty awesome. What did you have in mind for sharing the link?

m-bob, 2021-12-22:
Astonishing work: thank you! Seriously impressive. After watching the simulation, I visited your www.wokwi.com site and your GitHub site https://github.com/wokwi .

I also found your "club" site https://wokwi.com/club . I haven't (yet) joined, but I expect I will.

Do you have any plans to release local versions of your codes which wouldn't require web connectivity?

makermelissa, 2024-03-26:
Closing this issue for now since I'm not sure what to do with it.

closed 2024-03-26 by makermelissa