adafruit/Adafruit_ILI9341 #86: QT Py ESP32-C3 SPi compatibility

koogar, 2023-03-08:
- Arduino board:  QT Py ESP32-C3

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.19

Hi,

Just wondered if the ili9341 library is compatible with the QT Py ESP32-C3, as I have not been able to get an output.

I have been trying the same pins which I have been using on the QT PY ATSAMD21, plus some variations.
 
//QT Py ESP32-C3

// Libraries
//Espressif 2.0.7  no compile
//Espressif 2.0.6  no compile
//Espressif 2.0.5  compiles

//ili9341   1.5.12 compiles

//----------------------------------------------------------------------------------
#define TFT_CS   5
#define TFT_DC   7
#define TFT_RST  0 // also 9

/* hardware pins */
// TFT_SCLK  pin   8
// TFT_MOSI  pin   10

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST); // Use hardware SPI
//----------------------------------------------------------------------------------

Many thanks

Rupert

barnbrooksystems, 2023-03-24:
Having the same problem.  xiao works perfect (Same footprint & pinout) but same code does nothing on screen with the QT PY ESP32-C3 board

barnbrooksystems, 2023-03-24:
Thanks Rupert,

both the adafruit QT PY & Xiao C3 are a little fickle, I have ended up going back to Arduino IDE V1.8 rather that V2 for programming these.

I have tried PIN_D?, but still no change.  If i drop in a standard Xiao works straight away

koogar, 2023-03-24:
"I have tried PIN_D? "
I was referering to changing  5 to D5 etc,  I should  have also mentioned I was on arduino v1.8.19 Portable

koogar, 2023-03-24:
Try this on the XIAO C3.  All the latest library versions

https://github.com/koogar/Phat-Stats/tree/main/Experimental/XIAO_ESP32C3_ILI9431_GraphicsTest_PCB

barnbrooksystems, 2023-03-24:
Thanks I will try without change

Sorry I’m out of the office so Sent from my iPhone

https://github.com/koogar/Phat-Stats/tree/main/Experimental/XIAO_ESP32C3_ILI9431_GraphicsTest_PCB

barnbrooksystems, 2023-03-25:
Great works.  Thanks.  Now to find out what's different with my code

closed 2023-04-04 by koogar